Sakati84/3DPrintColorizer #8: Direct Drive extruder

Vantiveman, 2021-07-23:
I have a CR-10 that I have converted to direct drive (Micro-Swiss). The instructions for this plugin said it needed a larger retraction, but with direct drive where retraction is set to around 1.5mm a larger retraction is likely to cause a jam.
Is there a workaround or modification that would work with a direct drive converted Ender or CR-10?
Thanks,
Al

Sakati84, 2021-07-23:
Hey, the larger retraction is only needed for a bowden extruder because of the pressure in the bowden tube. I think for a direct extruder you can set it to zero or a small value. The worst thing that could happen is a little amount of oozing. In this case you can try out a ooze or draft shield. Please share your experience / parts when possible.

Vantiveman, 2021-07-23:
Cool! Thank for the reply. 

Seems like a nice thing to try as I keep switching back and forth between wanting dual extrusion or not. I’ll get printing over the weekend and order some sharpies. Might need a slight mod to the pen holder, as I have my reel holder mounter on the top bar of my CR-10.

If I get it installed and working I’ll post the experience and anything to look out for in a direct drive setting.

GrahamPrattWyalong, 2021-08-12:
Hi there, I have the same concerns with a DD extruder!

I have set it all up on the ender3.

all positions work fine and dandy except the retraction!

the retraction even if set to 0.0 and disabling retraction in the extruders in Cura themselves still rolls back the extruder by more than 75mm.. this is the amount of retraction needed to clear the drive cog!.

When I push the filament back in it oozes so much filament that it then causes x and y jamming!

I have looked in the script and in theory it inserts 0 for the retraction moves as best I understand it!

Any insight would be greatly appreciated!

BTW I have modified the STL for the holder and pen caps to allow for the DD (I manually cut the holder chunk so can't really provide the STL mods for that one!.

Regards.

Graham.

GrahamPrattWyalong, 2021-08-13:
OK I have run a bucket load of tests..

The only way this works for my DD on Ender3 without spitting anything out, Oozing or stringing is to....

Turn OFF retro in ALL pen extruders.
Turn ON retraction for Extruder 1 and set to normal anti stringing level
Ignore the Extra Retraction setting.
Rem out all inserted retraction and feed lines that show G1 F1500 E(- or nothing for return of filament)(Value set in box)
For me however modifying the script to still insert the lines but put a rem (;) in the output so they are ignored is best simple bet for now!

Hope this research helps others with Direct Drive.
I shall post the extended pen caps stiles to thingiverse as a remixed!

Regards.

Graham.

Sakati84, 2021-08-16:
Hi, thnaks for all the research! Can you post or do a pull request for you profile/script? I would like to have a direct extrusion Check box in the plugin.

GrahamPrattWyalong, 2021-08-17:
> Hi, thanks for all the research! Can you post or do a pull request for you profile/script? I would like to have a direct extrusion Check box in the plugin.

No Worries. here is a .2mm profile that works BUT..... there is a nasty extrusion occasionally that I have not nailed down yet! I am not sure why it occurs I am printing a more accurate pen holder before I test to try and find the bit of nasty code!

as regards your modified script, I simply inserted the rem where you add the extrusion movement + and - ergo "G1 F1500 E-" became ";G1 F1500 E-" and "G1 F1500 E"  became ";G1 F1500 E" as I am only using the DD machine.

I am happy to test your modified script if you want to send it to me!

oh and the .txt is a bodgy it is a .curaprofile file but the upload would not allow it in this forum so change it back when you get it!

[DD Colorizer Retract profile.txt](https://github.com/Sakati84/3DPrintColorizer/files/6996339/DD.Colorizer.Retract.profile.txt)

Cheers
G.

kohend, 2021-09-09:
I set it to 2 mm in my Prusa MK3S+, seems good, didn't get oozing despite using a very oozing filament.