Sakati84/3DPrintColorizer #9: PrusaSlicer support?

Sonosus, 2021-07-23:
I don't know if it's possible, but is porting this to PrusaSlicer viable? My Ender has really bad layer shifts with Cura. I'm not a developer so have no idea how difficult it would be, feel free to close this! 
Many thanks

Sakati84, 2021-07-23:
It‘s possible but to be honest I would like to wait for some results from the current Cura plugin. But if enough people want a Slicer Plugin I will definitely do it.

msazanov, 2021-07-26:
+1 PrushaSlicer (SuperSlicer)

Sonosus, 2021-07-30:
Ok, no bother. Off printing parts now, will give Cura another go.

adelyser, 2021-08-11:
+1 for PrusaSlicer

TechnomacherRyu, 2022-03-04:
+1 for PrusaSlicer

Dashrendar2507, 2022-04-04:
@Sakati84 Hi! Is this something you are still considering? I'm wanting to use this in SuperSlicer and wanted to know what were your plans (if any) before I start designing mounts for my machine. Thanks!

Nathan22211, 2022-04-17:
There's an advantage to using Prusa slicer/super slicer for this. You can color parts of the model for different extruders. I don't think you can do that in Cura

Niyazi-Oezel, 2022-12-19:
+1 for PrusaSlicer. I love Prusaslicer. Before I spent too much time with Cura, but for third party printers like the Ender 3 Prusaslicer just plug and play.

dbech, 2023-01-23:
I've not tried the project yet but I could imagine my way through what would needed to get this working with prusaslicer.
I'll take a crack at it later in the year.

scanlight, 2023-03-16:
would be awesome to have this function un prusa. left cura years ago, never looked back haha 
+1

LouisLe18, 2023-03-17:
YES +1 Prusaslicer

1412kauti, 2023-03-25:
+1 Prusaslicer

luemonkey123, 2023-05-09:
+1 Prusaslicer

Nathan22211, 2023-05-25:
for anyone needing prusa slicer support try this: https://github.com/krisrok/3DPrintColorizer/blob/feature/prusaslicer-compatibility/Software/PenColorizer.py seems to just recreate the Script class Cura uses but I haven't run a dry run with it yet. @Sakati84 this might be something to try perhaps

doggohusk, 2023-08-25:
+1 for PrusaSlicer

doggohusk, 2023-09-03:
i tried using that in prusaslicer and could not get it to workwhen i try to export it it just gives errors waying this class doesnt exist also theres no custom parameters like first pen position or pen offset so i cant even configure that