Sakati84/3DPrintColorizer #18: Outer wall only colorizing

ssilverm, 2021-08-03:
Would there be a way to only have the PenColorizer plugin generate gcode that colors the outer edges of the models?

I had an issue printing a dual extrusion Pikachu where because the sharpie colored in the whole tail layer the print separated.

[image]

Pineapplecake, 2021-08-03:
The setting `Infill > Infill Extruder` in CURA can be used to print all infill with Extruder 1, i.e. without colorizing.

Sakati84, 2021-08-03:
Unfortunatly no. You could set the top/bottom Extruder to 1 but than you would have no color there. To solve this you could: reduce the top/bottom speed or print it larger or change the model so it interlocks at this position

closed 2021-08-16 by Sakati84