Sakati84/3DPrintColorizer #24: MARLIN BUGFIX2.0.X

scootr1975, 2021-08-07:
I followed instructions and used the sample print. When I went to print it did not recognize the extruders. I use octoprint with my Ender 3 Pro. I have done several upgrades to the unit. I have the Creality 4.2.7 32bit board with direct drive extrusion. I have dual Z-axis. I have included some pictures Please help
[image]
[image]

Sakati84, 2021-08-07:
Hm.. for some reasons he doesn’t like the T0 command. This should normally refer to your first and only extruder. Does it work directly via USB or SD-Card?

stldenise, 2021-08-16:
I'm setting up an Ender 3 Pro also. I'm not a tech geek, so ignore if you did this: did you set up a new printer profile in OctoPrint with 7 nozzles? I was getting a similar error and they vanished when I set up a "new" printer.

scootr1975, 2021-08-17:
No I did not. Thank you for the information. Did you check the share button
when you added the extra extruders or did you put in the offsets there.

stldenise, 2021-08-17:
You know, it was late when I did that...didn't even see it! So, no I did not click shared nozzle or anything else.

scootr1975, 2021-08-17:
Haha, thanks!

closed 2021-08-19 by Sakati84

scootr1975, 2021-08-19:
Awesome, thanks!