Sakati84/3DPrintColorizer #28: Improvement Idea: Hang sharpies from the tip

ianespana, 2021-08-08:
First things first, I have checked out this system and I am quite impressed by how simple it is!

However there is one flaw with it: Because of the way the sharpies hang from the back with the end cap you effectively remove a chunk of printing space in Z. So, what about changing the way sharpies are held?
Instead of hanging them from the end it'd be better to hang them from the front, similar to adding a front cap to the sharpie. This would not only allow you to reclaim a lot of space in Z, but it would also allow you to use different length sharpies.
It'd also mean you have a more secure way of holding the sharpie as it's own weight would be pushing it into the cap.

Depending on the mounting solution it might even be capable of mounting different types of markers while keeping the pen arm and the rack the same.

Has this been considered or tried before? What do you think some drawbacks with this system might be?

Sakati84, 2021-08-09:
Hi, thanks for your suggestion. I'm not 100% sure I understood it right. From what I understood it seems like this would compress the bowden cable to much.. Could you perhaps provide a very basic drawing?

ianespana, 2021-08-09:
Sure thing!
[image]

On the left is the current configuration where the end cap is what holds the pen in place
On the right is my proposed change, where the cap goes on the front of the pen and reduces the length that's hanging down

RyushoYosei, 2021-08-10:
The issue I see with this, is you would need to have the holder lower, or a Z axis that could actually raise high enough to get to it.

Sakati84, 2021-08-10:
I see some problems with bowden extruders. I will leave this open as an idea. I'm working on a new holder/rack design which should solve a lot of this problems. If it doesn't turn out to work I will revisit this idea.