Sakati84/3DPrintColorizer #31: Missing source code

mousethief, 2021-08-11:
I notice that none of the .stl object code files for the 3D printed parts have any corresponding source code files anywhere on the repo. How are people even making contributions?

ShadowBr0ther, 2021-08-11:
Those files are located unter the folder "Hardware"...

mousethief, 2021-08-11:
There are only object (stl) files in the directory 'Hardware'.

csanii123, 2021-08-21:
If you go 1 directory up you will find a Software directory. In there is the Python code that does the post processing of the gcode to do the coloring.

kohend, 2021-09-09:
I just edited the STLs, since the calibration is based on the actual hardware setup, so dimension differences in mount and caps do not seem to matter as long as you're calibrated.