Sakati84/3DPrintColorizer #36: seems to be over retracting without marker

jmerc77, 2021-08-12:
when there's no marker an its extruding the retracts are longer then the detracts. may vary on the retraction during painting value. still trying to find exact cause...

jmerc77, 2021-08-12:
retraction during painting = 0 did not help this.

Sakati84, 2021-08-12:
Hi, there was a new pull request for the cura Profile which should fix this. You can redownload and Import the Profile or just disable "retract on layer change" in cura.

jmerc77, 2021-08-12:
tried turning off "retract on layer change" but still having the problem. also there are z-hops when z-hop is disabled. another problem is M900 for linear advance on each tool. should be only one tool.
[gcode.txt](https://github.com/Sakati84/3DPrintColorizer/files/6977194/gcode.txt)

jmerc77, 2021-08-12:
oh I forgot to mention, sometimes it would print slow but sometimes fast. varying feedrate?

jmerc77, 2021-08-12:
hay! randomly started working. I think it was the wall count?? the M900 is still a problem though. I'll post separate issue. although not effecting print quality so far and no warnings.

closed 2021-08-12 by jmerc77