Sakati84/3DPrintColorizer #39: minor bug... M900 for linear advance on each tool.

jmerc77, 2021-08-12:
M900 for linear advance on each tool. should be only one tool.
what I have in gcode:

`...
M900 K0.130000 T3 ;added by LinearAdvanceSettingPlugin
M900 K0.130000 T0 ;added by LinearAdvanceSettingPlugin
M900 K0.130000 T2 ;added by LinearAdvanceSettingPlugin
M900 K0.130000 T1 ;added by LinearAdvanceSettingPlugin
M900 K0.130000 T4 ;added by LinearAdvanceSettingPlugin
...`

should be:

`...
M900 K0.130000 ;added by LinearAdvanceSettingPlugin
...`

Sakati84, 2021-08-16:
I don´t really see the problem here - this was generated by another plugin. How should this be fixed in the colorizer plugin?

jmerc77, 2021-08-16:
the T parameters should not be there. mainly for consistency. even better would be to have only one of those left in but without the T parameter. same should be filtered for other tool specific G and M codes. even though this may not happen to everyone it is still a good idea to cover all scenarios.

Sakati84, 2021-08-16:
What does this look like with the LinearAdvanceSettingPlugin deactivated?

jmerc77, 2021-08-16:
the developer of that plugin did not provide a disable box. I can only uninstall it or set values to 0. without the plugin there would not be such lines. however, I use the plugin to help with making my line width more consistent and uninstalling it would likely hurt my print quality. setting to 0 has no effect on the existence of these lines. also the plugin is a market place plugin that loads on startup. therefore in a way not a post processing plugin. 
also not quite sure if it is fixable as I don't know which runs first. hopefully the linear advance plugin (and others like it) run first.

Sakati84, 2021-08-19:
Sorry but I won´t & can´t fix other plugins.

closed 2021-08-19 by Sakati84