Sakati84/3DPrintColorizer #41: Need Test Routines and a little feedback

ridan, 2021-08-12:
This is such a cool project. There are a couple of things that need to be included in the documentation to speed up the process of installing, tuning and getting started.

**_Testing routines_**
Firstly there needs to be a quicker way of testing without starting a print. I wonder if this can be a set of scripts that we load next to the PenColorizer.py
these scripts can do a few things
1) pick pen X, where X is one of the extruders. This test will allow us to see if the printer can pick the pen in the right direction or not and find out if any of them are misaligned
2) Draw a circle on X layer, here X is the layer height. This will have a checkbox called "assume pen loaded". This will allow us to see if z is properly calibrated for the printer. make a quick circle or some shape at a certain layer. With the checkbox we can combine a test of picking up a pen and testing the Z for coloring

Also if there were a few instructions on what kind of pen holder arm is needed. I just realized it is best to have the pen holder be much further out from print head. If not then it always hit the other pends when going up. I designed my own because I have a satsana fan shroud and at one point it seems like the pen was going to get stuck in the fans and ruin it. Most of the modders that use this will have custom fan shrouds with exposed fan so be careful. 

apart from that it's a great project

Sakati84, 2021-08-16:
Good ideas. I still have some other stuff to do but after this I will try to add this. As for the position of the arm: I recently updated the Wiki for the BLTouch holders and there is a description on where to put the arm. I know one other person who had this problem before.

csanii123, 2021-09-04:
Basic pen test files for #1 that you mentioned.
[Pen_tests.zip](https://github.com/Sakati84/3DPrintColorizer/files/7110133/Pen_tests.zip)