Sakati84/3DPrintColorizer #43: Cant Print Second Layer

Shynka, 2021-08-16:
I modified my GEEETECH A10M 3D Printer.  removed one of the extruders repositioned the other one with double sized tape on the  power supply. Need to reposition the fan. But after doing so much testing and not enough screws I hot glued everything together, leaving me with 4 colors total...

I've probably tried for a full day. But for the life of me I can not get the second layer to stick the Cura.
I've leveled the bed so many times.
I have normal prints working with Repetier and Easyprint Lite Release with the profiles provided by the Company that sells the printer.

So anyone know how to Either change the profile to another software or how to get Cura to work.

Sakati84, 2021-08-16:
Did you use the provided Profile?

Shynka, 2021-08-16:
> Did you use the provided Profile?

I tried the provided one. Looking into the code and and comparing with the that comes with the printer.
[image]

Sakati84, 2021-08-19:
What is your print speeds for the top/bottom layers & walls?

Shynka, 2021-08-20:
[image]
[image]
Like it gets the first layer great. then the second is not sticking