Sakati84/3DPrintColorizer #44: Can't Reach Pen #7

JohnSpoAA, 2021-08-18:
I ran a test print to ensure all pens could be reached on a standard print, Pen #1 aligns perfectly, but there is more and more of a gap the farther down the rack the hotend goes. Pen #7 Doesn't align at all with the pen holder.

My steps/mm are calibrated and the size is correct, as there should be what seems like 40mm between the pen rack holes.

Could I suggest some sort of averaging system?

Take the first and last hole measurements and average them out to equal 7 equally separate holes?

I'm really grasping at straws here, it seems like I can't find this issue anywhere else. 

Thank you in advance!

HunterG6700, 2021-08-23:
Try to loosen up the screw at 7 and adjusting it a little until it works. That's what I did and mine seems to work so far. Also maybe make your first position x 1mm larger

ivanmailru, 2021-09-03:
Same problem here. I have Ender 3V1 with no upgrades and stock firmware. The slicer used is Cura 4.10

csanii123, 2021-09-04:
Just eyeballing the centers for the pens using Cura and I see about 42mm for 2-3, 4-5, and 6-7 and about 27mm for 3-4 and 5-6.