Sakati84/3DPrintColorizer #62: My nozzle is digging on my print

CjSd1, 2021-08-29:
The first layers will be laid fine but later on it will be pushing against the print and flexing the springs of my bed, and grind against the print.

Sakati84, 2021-09-07:
Pls send me your GCode so I can make sure it is a Software and not a Hardware porblem.

MXZZ, 2021-09-09:
I have the same issue, seem that using the ender3 pro and the "Custom FDM Profile" the Z axis appear to be reversed
here is my gcode
[CFFFP_DualStrusion-spiralCylinder_m1.zip](https://github.com/Sakati84/3DPrintColorizer/files/7135569/CFFFP_DualStrusion-spiralCylinder_m1.zip)

JMVS, 2021-09-13:
I'm having the same problem on an Ender 3 v2. I have to manually increase by 0.20 the Z Offset for every new layer to prevent crashing over printer parts. Here's the code:
[CFFFP_Tolve_Deco_Llavero_(Italia).zip](https://github.com/Sakati84/3DPrintColorizer/files/7150550/CFFFP_Tolve_Deco_Llavero_.Italia.zip)

MXZZ, 2021-09-13:
how do you increase the z offest ? you replace the gcode lines by hand or you have a script?

MXZZ, 2021-09-13:
could it be that our Z axis is wired wrongly? 

I noticed that im able to print normally without any colour so i can exclude the profile/printer i'm trying to identify the issue in the python script ..

JMVS, 2021-09-13:
> 
> 
> how do you increase the z offest ? you replace the gcode lines by hand or you have a script?

In the Tune menu, whilst the printer is printing.

> 
> 
> could it be that our Z axis is wired wrongly?
> 
> I noticed that im able to print normally without any colour so i can exclude the profile/printer i'm trying to identify the issue in the python script ..

The printer is doing just fine with regular prints and the finish is far superior in those cases. I've only starting using 3DColorizer  recently and this was my second colored print after the first test (which did something similar but wasn't as noticiable as it was lower -less layers-).

Probably something to do with the Python script and maybe a combination with the printer (Ender 3v2) / profile.

Let's see if @Sakati84 can come up with something regarding this.

hfgd123, 2021-11-05:
Same problem here

Sakati84, 2021-11-10:
@MXZZ your GCode looks fine to me. I would guess your Z Axis misses a step because of high speed. Can you try printing with a lowered Z Axis speed?

dschaadt, 2022-03-28:
Look like a cool project, unfortuantely it's not working for me on an Ender 3 V2, even with z axis speed set to 5. Has this been resolved? Where can I add something that the z axis moves up for each printed layer?

MXZZ, 2022-06-25:
I commented this line in the python script 
            '''lines = self.addExplicitZ(lines)'''
and it seems to work so far ( i'm trying a simple model) ,   please try and let me know, I'm using now a Voxelab Aquila x2 ( it had the same issue)

fnorbenden, 2023-02-07:
> I commented this line in the python script '''lines = self.addExplicitZ(lines)''' and it seems to work so far ( i'm trying a simple model) , please try and let me know, I'm using now a Voxelab Aquila x2 ( it had the same issue)

This also fixed an issue I had with height being added to layers and breaking the print. Not sure what that function does (I haven't had a chance to look at it) but it seems to just break things.

fnorbenden, 2023-02-15:
I took another look at the function itself and it's trying to use next() on a list (from re.findall) instead of an iterator. That is probably breaking the output and leading to unpredictable results messing with Z coordinates of G0 and G1 commands.

I also don't see any real reason to generate the list in the first place.

`coord = re.findall(r'[XYZF].?\d+(?:.\d+)?', line)
                Z = next((i for i in coord if i.startswith('Z')), None)`

and

`Z = re.findall(r'Z.?\d+', line)`

should return the exact same thing, the Z coordinate for a G0/G1 command, which is the only thing we care about. Without the typing issue.

I'll have to test it