Sakati84/3DPrintColorizer #66: [Question / Feature Request] Drawing / Coloring on paper without hotend / heated bed

JMVS, 2021-09-03:
Great project! Just finished printing all the parts, installing, calibrating and doing a first test run on an Ender 3 v2. All seems to be working according to plan. 

I would like to know if it is possible to use the system to just _draw_ on paper without using the hotend nor the heated bed. I assume is possible by just using a single layer model, not using Extruder 1 and turning off hotend & bed temperatures and setting the fans off. I'm not really sure about the Z-Offset and maybe a Z-Hop when travelling?

I would like to hear from your thoughts about this.

**Thanks again for this project!**

JMVS, 2021-09-04:
### Alright first test drive with the butterfly went OK:
[image]
_(don't mind the circles, they were the very first try)_
Took really loooong time (about an hour) because the Extruder 1 (hotend) did all the paths first at initial layer speed. Also forgot to turn on Z-Hop for all extruders.

### Second test went really well:
[image]
[image]
This took about 15 minutes as I manually deleted all the Extruder 1 (hotend) lines from G-Code. I've yet to figure out if this can be done automatically done by the post-processing plugin.

### Color setup:
[image]

### What I changed:
**Profile**:
- Disble all heating and cooling for all extruders (basically turning off or setting to zero)
- Enable Retraction on and set retraction distance to 0 for all extruders
- Enable Z-Hop and set distance to 5.0mm to all extruders
- Changed speed up
- Some other tweaks here and there (0% infill, spiralize outer countour, and some others)

**PenColorized Post-Processing Plugin**:
- Retraction during painting set to 0
- Disable Interlace colors

**G-Code**:
- Manually removed all Extruder 1 (hotend) related lines.

Test file (rename extension .gcode)
[CFFFP_Mariposa_rev2.txt](https://github.com/Sakati84/3DPrintColorizer/files/7110311/CFFFP_Mariposa_rev2.txt)
**_(NOT for printing your printer values might differ and cause problems)_**

Here's the Illustrator file I used:
[Butterfly test.zip](https://github.com/Sakati84/3DPrintColorizer/files/7110321/Butterfly.test.zip)

Sakati84, 2021-09-07:
Very cool! I think I will add a "Draw only"-mode to the plugin in the future!

JMVS, 2021-09-07:
What is very important is to make sure there's **no extruder movement**. In my tests I took out the filament just to be on the safe side, as moving the extruder (either way) with the hotend cold (and thus the filament) would be a bad idea.

Some other things I noticed for this to work perfectly:
- When changing pens, the pen goes to the last "printed" point and makes a dot. Then moves up and travels to the next point (where the new pen should start drawing)
- The profile should be tweaked so there's only one wall and the rest infill).
- Following this, I don't know how I messed up, but from one point forward I started having a lot of Z Hops on the same pen (maybe due to the thickness of the drawing the settings on the profile).
- Not much of a problem, and maybe it's to accomodate different printers, but there's too much Z travel when putting one pen and taking another. This could be set as a parament of the plugin so can be adjusted by the user.
- All extuder 1 movements should be removed from gcode
- All temps & fans should be at zero / off

I'm sending some other sample over cardboard (had to adjust Z-Offset):
[image]

I can help with testing if needed. Thanks again for a great work!