Sakati84/3DPrintColorizer #68: Color offset from print

buddybu, 2021-09-07:
I am seeing an offset in both the x and y axis, but mostly x.  I did the calibration as described, is there something I am missing?
[image]
Ignore the bad print, I am working through clogging issues.  I am running on anet a8 hardware, but with an E3Dv6 clone hotend and Bowden extruder as well as an SKRv1.4

buddybu, 2021-09-07:
[configuration.zip](https://github.com/Sakati84/3DPrintColorizer/files/7118793/configuration.zip)

Sakati84, 2021-09-07:
Sorry but I don´t have any clue about this. The generated GCode for coloring isn´t diffrent from the other GCode generated by Cura so I don`t see anything special that could go on here. Perhaps you can manually look at the GCode and make sure it´s alright?

buddybu, 2021-09-08:
The solution is to add your X_MIN_POS/Y_MIN_POS or MANUAL_X_HOME_POS/MANUAL_Y_HOME_POS to the X and Y offset in the post processing dialog.  Be sure to use the absolute value, i.e., if your X_MIN_HOME_POS is -18.4, then add 18.4 to the X offset you discovered lining up against the screw, and do the same for the Y offset you discovered lining up against the screw.