Sakati84/3DPrintColorizer #69: Suggestion: have script skip painting over the infill

christopherjamesmeyers, 2021-09-12:
I've noticed the script spends a lot of time painting over infill that wouldn't normally be seen and causes issues with layer adhesion (perhaps not letting the ink dry)

Sakati84, 2021-09-14:
Normally it should only paint the walls. Can you send me a photo of the infill painting?

christopherjamesmeyers, 2021-09-25:
Realized I was only looking at the first few layers before it started the infill. however, there is still the issue of layer adhesion

csanii123, 2021-09-25:
I didn't have any issues with my basic multi-color prints and layer adhesion. Do you have any issues when doing non-multi-color prints ? Perhaps you don't have your bed close enough and thus not enough "squish" between layers.

Sakati84, 2021-09-25:
What is your top/bottom  speed?

christopherjamesmeyers, 2021-09-26:
@csanii123 regular prints work just fine
@Sakati84 15 mm/s