Sakati84/3DPrintColorizer #70: Printing pen rack

hfgd123, 2021-09-19:
I wanted to print the pen rack, but the first layer was only printed partwise and the next layers (until I canceled the print) the hotend scratched over the lower layers. Everything else I try to print works well. So I thought maybe anyone here could help me. Heres a picture of what was intended to be the pen rack (printed with an Ender-3) :
[image]

csanii123, 2021-09-19:
I haven't tried printing this model but it looks like your bed isn't level as the bottom right seems to have no brim/raft while the top does seem to have it intact. Suggest a relevel and use a skirt and see if that sticks well. If it does then move back to printing with a brim.

hfgd123, 2021-09-19:
The problem with that: I leveled the bed a few minutes before I started printing this. And the first time I tried printing the pen rack (also with leveling right before), there was the same problem. But I'll try printing with a skirt next weekend

csanii123, 2021-09-19:
Did you heat the bed/nozzle and then level ? When I 1st got my printer I wasn't doing that and prints were "ok" but the stick to the bed was hit and miss. Since I heated the bed/nozzle and then leveled...prints are much better and stick to the bed very well.

hfgd123, 2021-09-19:
Bed and nozzle are always heated when I am leveling, yes

hfgd123, 2021-09-26:
It worked now. thx

closed 2021-09-26 by hfgd123