Sakati84/3DPrintColorizer #78: after slicing the model doesnt merge , but shows normal in slice sim. not gcode4.

howy123, 2021-10-30:
I followed all instructions and it does work BUT doesnt slice right. When i slice a multi color model in cura of coarse, it slices fine i can set any extruder i want and it slices the way it should but after i put the file on sd I saw the model was split up again in random places even tho I checked the slice in the sim and its fine. when i look at g-code its split up. why is this ? I see others on youtube using this mod and it works. It retrieves pen just fine , just does something to gcode wrong . Ive tried reinstalling cura 4.11.0 and same issue still.
any help please.......

Sakati84, 2021-11-02:
What you see in the preview are probably the gcode paths for the pens. Did you try printing the model?

howy123, 2021-11-02:
I played with the offsets and it looks good I haven't had time to test it
yet. It was doing a weird thing it would split my sliced model in to 2
parts. I am starting the settings on 0 and work my way to the correct
offset cause my measurements were prob not right. Thank you for the
reply.....