Sakati84/3DPrintColorizer #91: Not fetching marker

GoldenSnitch42, 2022-03-03:
I'm new to 3D printing so I apologize if I'm making a newbie mistake.  I'm using the stock firmware on my Ender 3 v2.  I've followed the instructions as closely as I could and I think I have everything setup correctly.

I did the Cura setup and sliced the multicolor parrot linked in the instructions after assigning the parts to the 4 extruders. 

When I print I get the first layer or so down and the printing pauses for a few seconds without moving the head and then it continues to print.  If I let it go this happens several more times.  I've let it print a couple of millimeters but as it never attempted to pickup a marker I stopped it.  I'm assuming that at those pauses it's supposed to go get the marker and then do the coloring.

I've setup OctoPrint and have attempted the same print from it and I get many error messages all something like "Not queuing T0, that tool doesn't exist according to the printer profile or was reported as invalid by the firmware. Make sure your printer profile is set up correctly."  Sometimes its for T1 or T2 as well.

Should this work with the stock Creality firmware (1.0.2) or do I need to build and flash my own Marlin firmware?

Any other suggestions for getting this to work?

csanii123, 2022-03-03:
I was able to make this work on my stock Ender 3 v2 with firmware v1.0.2
You need to follow the software setup instructions as that inserts the 
pen pickup/return commands.
https://github.com/Sakati84/3DPrintColorizer/wiki/4-Software-Setup

csanii123, 2022-03-03:
> Thanks, it's good to hear someone has gotten it to work with my setup.
> I followed those instructions but I'm going to start over and do them
> again to make sure I didn't mess something up. Have you used this with
> OctoPrint? If so do you get any errors when doing so?

I have not tried with Octoprint.
I copied the file direct to an SD card and used that.
In theory it should work the same for Octoprint as all it is doing is 
sending the GCODE commands 1 line at a time the same the printer reading 
from the SD card.

I should clarify I did not try printing the parrot but created my own 
multi-color print for testing.

GoldenSnitch42, 2022-03-06:
I got it to work, I redid all the software parts and everything is good to go the second time around.  Parrot printed pretty well.  I think my x axis offset might need tweaking.

Another question regarding the Ender 3 v2, the instructions say to set the Z axis speed to 100.  The max I can set it to is 10 from the default 5.  Is that the best I can do?

csanii123, 2022-03-06:
I honestly never played with the max z-speed as I wasn't sure if it 
would cause issues with my printer.

With a non-stock firmware you might be able to adjust further but I am 
sure Creality set a max for a reason.

I tried lowering the penholders and that was "ok" but still slow for 
multi-color printing.

There was a limit on how low the pens could be since the pen swap 
routine had to account for enough clearance to make the swap.

The other idea I was thinking about was a rotary pen holder so that the 
pen swap didn't have to go to the top but rather would just go to the 
far left or far right to swap pens.

I didn't actually build something for this as it would also require new 
code for the amazing work that the current solution provides.