Sakati84/3DPrintColorizer #109: Stops printing

ElijahSpirit, 2023-02-10:
After printing the first layer the extruder temperature suddenly drops to 0. it nevers moves up or anything. just keeps printing like nothinbgs happened but isnt moving extruder gear and sets the temperature to zero

ElijahSpirit, 2023-02-11:
It is a direct drive ender 3.

fnorbenden, 2023-02-11:
The script should never change the hotend temperature; the only time it touches M104 is specifically to remove them to avoid firmware issues with a cold "extruder" that's been changed to a pen.

Have you tried slicing/printing this model without the script (remember to screenshot the settings if you remove it because it is dumb about keeping them)? Could you pastebin your gcode for each? Are you using marlin or klipper?

ElijahSpirit, 2023-02-11:
Hi! The issue occurs when using octoprint with an error tracking default plugin.

closed 2023-02-11 by ElijahSpirit