Sakati84/3DPrintColorizer #111: plugin lot loading in Cura

scanlight, 2023-03-16:
Copied to file to the plugin folder stated in the instruction and restarted Cura. still the plugin doesnt show up in the plugin list.
Tried cura 4.9.0, 4.9.1 5.2 and 5.3

Is this script version specific or some other issue ?

LouisLe18, 2023-03-17:
I used 4.9.0 and it loaded. But I'm still trying to get it to print properly

scanlight, 2023-03-17:
checking the logs it has an issue with the title starting with a decimal. changed this and tried again. 
now issue at line 8
<!DOCTYPE html>
  ^
 SyntaxError: invalid syntax

scanlight, 2023-03-17:
full Log part with error:

2023-03-17 19:27:53,229 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]: Exception: Exception occurred while loading post processing plugin: invalid syntax (PenColorizer.py, line 8)
2023-03-17 19:27:53,229 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]: Traceback (most recent call last):
2023-03-17 19:27:53,230 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:   File "C:\Program Files\UltiMaker Cura 5.3.0\share\cura\plugins\PostProcessingPlugin\PostProcessingPlugin.py", line 201, in loadScripts
2023-03-17 19:27:53,230 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:     spec.loader.exec_module(loaded_script)  # type: ignore
2023-03-17 19:27:53,230 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:   File "<frozen importlib._bootstrap_external>", line 879, in exec_module
2023-03-17 19:27:53,231 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:   File "<frozen importlib._bootstrap_external>", line 1017, in get_code
2023-03-17 19:27:53,231 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:   File "<frozen importlib._bootstrap_external>", line 947, in source_to_code
2023-03-17 19:27:53,231 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
2023-03-17 19:27:53,232 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:   File "C:\Users\robbert\AppData\Roaming\cura\5.3\scripts\PenColorizer.py", line 8
2023-03-17 19:27:53,232 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:     <!DOCTYPE html>
2023-03-17 19:27:53,232 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:     ^
2023-03-17 19:27:53,233 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]: SyntaxError: invalid syntax
2023-03-17 19:27:53,234 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]: Exception: Exception occurred while loading post processing plugin: invalid syntax (PenColorizer.py, line 8)
2023-03-17 19:27:53,235 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]: Traceback (most recent call last):
2023-03-17 19:27:53,235 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:   File "C:\Program Files\UltiMaker Cura 5.3.0\share\cura\plugins\PostProcessingPlugin\PostProcessingPlugin.py", line 201, in loadScripts
2023-03-17 19:27:53,235 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:     spec.loader.exec_module(loaded_script)  # type: ignore
2023-03-17 19:27:53,236 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:   File "<frozen importlib._bootstrap_external>", line 879, in exec_module
2023-03-17 19:27:53,236 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:   File "<frozen importlib._bootstrap_external>", line 1017, in get_code
2023-03-17 19:27:53,236 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:   File "<frozen importlib._bootstrap_external>", line 947, in source_to_code
2023-03-17 19:27:53,237 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
2023-03-17 19:27:53,237 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:   File "C:\Users\robbert\AppData\Roaming\cura\5.3\scripts\PenColorizer.py", line 8
2023-03-17 19:27:53,237 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:     <!DOCTYPE html>
2023-03-17 19:27:53,238 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]:     ^
2023-03-17 19:27:53,238 - ERROR - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [221]: SyntaxError: invalid syntax

scanlight, 2023-03-17:
solved! thnks

closed 2023-03-17 by scanlight