Sakati84/3DPrintColorizer #120: [Suggestion] Compatibility with newer models in ender 3 line up

doggohusk, 2023-08-28:
printed everything in hardware folder for my ender 3 max neo which i think released earlier this year and noticed the penrack doesnt go low enough for the print head to put the marker back up even putting the rack on the bottom part of the top bar on my printer it was still too high managed to find an stl that would get let it go low enough to (hopefully) let it put the marker back up on the rack, still printing as off right now but would be helpful if there were available compatable pen racks for later ender 3 series printers so as to not waste filament.

doggohusk, 2023-08-28:
finished printing today in the morning just installed couple of hours ago rack is low enough now so it works great

thisnyjared, 2024-01-07:
It is compatible with any bl touch compatible printer except for corexyz is that what you mean?